riscv-mcu/riscv-gnu-toolchain #3: How to compile POSIX source?

tuupola, 2020-10-24:
Not sure if this is the correct place, but at least the branchname `nuclei-multilib`  suggests the Nuclei RISC-V toolchain. I am unable to figure out how to compile POSIX source. I have the following code:

```c
#include <time.h>

int main ()
{
    struct timespec time;
    clock_gettime(CLOCK_MONOTONIC, &time);
    return 0;
}
```

Which compiles fine with with `-std=gnu99` option on Linux.

```
$ gcc -std=gnu99 main.c 
```

With the RISC-V toolchain compilation fails. It feels like `-std=gnu99`  is ignored and `-std=c99` is used instead.

```
$ riscv-nuclei-elf-gcc --version
riscv-nuclei-elf-gcc (GCC) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ riscv-nuclei-elf-gcc -std=gnu99 main.c 
main.c: In function 'main':
main.c:7:5: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
    7 |     clock_gettime(CLOCK_MONOTONIC, &time);
      |     ^~~~~~~~~~~~~
main.c:7:19: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
    7 |     clock_gettime(CLOCK_MONOTONIC, &time);
      |                   ^~~~~~~~~~~~~~~
```

Should the RISC-V toolchain support POSIX extensions or am I missing something obvious?

fanghuaqi, 2020-10-27:
Maybe it is caused by the elf toolchain is prepared for embedded system with newlib, but if you want to compile with posix extension, maybe you need glibc.

fanghuaqi, 2020-10-27:
You can find `CLOCK_MONOTONIC` defined in https://github.com/riscv/riscv-newlib/blob/riscv-newlib-3.2.0/newlib/libc/include/time.h#L287-L295, which require `_POSIX_MONOTONIC_CLOCK` to be defined, but it also required another header file https://github.com/riscv/riscv-newlib/blob/riscv-newlib-3.2.0/newlib/libc/include/sys/features.h#L441, where __CYGWIN__ should be defined to enable this  `_POSIX_MONOTONIC_CLOCK` macro. 

So if you still need to compile code like this you can pass:

`riscv-nuclei-elf-gcc -std=gnu99 -D_POSIX_MONOTONIC_CLOCK=200809L main.c`

Thanks
Huaqi

tuupola, 2020-10-27:
This will only define `CLOCK_MONOTONIC`  const but still does not include the `clock_gettime()` function. 

```
riscv-nuclei-elf-gcc -std=gnu99 -D_POSIX_MONOTONIC_CLOCK=200809L main.c
main.c: In function 'main':
main.c:6:5: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
    6 |     clock_gettime(CLOCK_MONOTONIC, &time);
      |     ^~~~~~~~~~~~~
/opt/nuclei/gcc/bin/../lib/gcc/riscv-nuclei-elf/9.2.0/../../../../riscv-nuclei-elf/bin/ld: /tmp/ccq0zQoY.o: in function `main':
main.c:(.text+0x10): undefined reference to `clock_gettime'
```

My understanding user is not really supposed to pick single features through defines. To enable full POSIX.1-2008 specification one would either use `-std=gnu99` or the `_POSIX_C_SOURCE` feature request macro:

```
$  riscv-nuclei-elf-gcc -D_POSIX_C_SOURCE=200809L main.c 

main.c: In function 'main':
main.c:6:5: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
    6 |     clock_gettime(CLOCK_MONOTONIC, &time);
      |     ^~~~~~~~~~~~~
main.c:6:19: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
    6 |     clock_gettime(CLOCK_MONOTONIC, &time);
      |                   ^~~~~~~~~~~~~~~
main.c:6:19: note: each undeclared identifier is reported only once for each function it appears in
```

Which does not fix the problem. Reading through source I came to conclusion that `clock_gettime()` is not implemented. I also tried using more portable `clock()` but it always returns `-1`  suggesting that `_times()` is not implemented.

https://github.com/riscv/riscv-newlib/blob/riscv-newlib-3.2.0/newlib/libc/time/clock.c#L58-L62

I did a naive implementation of  `_times()` but I think it belongs to https://github.com/Nuclei-Software/nuclei-sdk so better continue this discussion there.

closed 2020-10-28 by tuupola