riscv-mcu/riscv-gnu-toolchain #13: Nuclei Toolchain 2023.10 failed to disable V-ext autovectorization

bigmagic123, 2024-02-27:
When using the rvv extension compiler, if you do not enable rvv automatic vectorization, there is still a chance that rvv instructions will be generated.

[image]

bigmagic123, 2024-02-27:
For more detailed information, please refer to the link

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=112538

closed 2024-03-12 by fanghuaqi

fanghuaqi, 2024-07-01:
And from `2024.06` release, `-mmemcpy-strategy=` option in 2023.10 changed to `-mstringop-strategy=` in 2024.06, see https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=3dc7947e5d766ed2f30c9aa8286b36f9287a1d80, and Nuclei SDK also changed this option, see https://github.com/Nuclei-Software/nuclei-sdk/commit/a487143e9d09757ef2613fee10440f5dc00bfb36